sipeed/Maixduino #2: esp8285 does not respond

GitJer, 2019-04-09:
I have tried the WiFiEsp library examples WebClient and WebServer. After indicating that the Maix GO has hardware serial, defined in sipeed_go.h, they compiled. With a logic analyzer on pins 6 and 7 I verified that the esp8285 gets the correct signal (AT\r\n). But the esp8285 does not respond and the example concludes that it is not active. Is there something I should do to make the esp8285 active?

btx000, 2019-04-09:
You need to try setting HIGH on pin 8 to enable esp8285.

GitJer, 2019-04-09:
Thank you!
It now gives a response, but not 'OK'. It responds with 0x42 0xFF. So the example still concludes that the ESP cannot be initialized. If I give the AT command to reset, the ESP gives responses, but again not what is expected.

Also if I try the [maixpy demo](https://github.com/sipeed/MaixPy_scripts/blob/master/basic/demo_net_socket.py), with correct ssid and passwd, I get an error:

>  OSError: couldn't init nic esp8285 ,try again please

Maybe there is something wrong with the esp8285 on my board?

btx000, 2019-04-10:
Try using logic analyzer to read the serial output when esp8285 first startup.

GitJer, 2019-04-11:
During startup it seems to respond correctly:

At first the baud rate is 74880:

\r \n ' ' e t s ' ' J a n ' ' ' ' 8 ' ' 2 0 1 3 COMMA r s t ' ' c a u s e : 2 COMMA ' ' b o o t ' ' m o d e : ( 3 COMMA 0) \r \n \r \n l o a d ' ' 0 x 4 0 1 0 0 0 0 0 COMMA ' ' l e n ' ' 2 5 9 2 COMMA ' ' r o o m ' ' 1 6 ' ' \r \n t a i l ' ' 0 \r \n c h k s u m ' ' 0 x f 3 \r \n l o a d ' ' 0 x 3 f f e 8 0 0 0 COMMA ' ' l e n ' ' 7 6 4 COMMA ' ' r o o m ' ' 8 ' ' \r \n t a i l ' ' 4 \r \n c h k s u m ' ' 0 x 9 2 \r \n l o a d ' ' 0 x 3 f f e 8 2 f c COMMA ' ' l e n ' ' 6 7 6 COMMA ' ' r o o m ' ' 4 ' ' \r \n t a i l ' ' 0 \r \n c h k s u m ' ' 0 x 2 2 \r \n c s u m ' ' 0 x 2 2 \r \n \r \n 2 n d ' ' b o o t ' ' v e r s i o n ' ' : ' ' 1 .  7 ( 5 d 6 f 8 7 7) \r \n S P I ' ' S p e e d ' ' : ' ' 4 0 M H z \r \n S P I ' ' M o d e ' ' : ' ' D O U T \r \n S P I ' ' F l a s h ' ' S i z e ' ' & ' ' M a p : ' ' 8 M b i t ( 5 1 2 K B + 5 1 2 K B) \r \n j u m p ' ' t o ' ' r u n ' ' u s e r 1 ' ' @ ' ' 1 0 0 0 \r \n \r \n r f ' ' c a l ' ' s e c t o r : ' ' 2 5 1 \r \n f r e q ' ' t r a c e ' ' e n a b l e ' ' 1 \r \n r f [ 1 1 2 ] ' ' : ' ' 0 3 \r \n r f [ 1 1 3 ] ' ' : ' ' 0 0 \r \n r f [ 1 1 4 ] ' ' : ' ' 0 1 \r \n \r \n S D K ' ' v e r : ' ' 2 .  2 .  1 ( 6 a b 9 7 e 9) ' ' c o m p i l e d ' ' @ ' ' J u n ' ' ' ' 7 ' ' 2 0 1 8 ' ' 1 3 : 5 9 : 1 4 \r \n p h y ' ' v e r : ' ' 1 1 3 6 _ 0 COMMA ' ' p p ' ' v e r : ' ' 1 0 .  2 \r \n \r \n

Then it switches to 9600:
\r \n r e a d y \r \n

But when it receives the 'AT\r\n' it still responds with 0x42 0xFF.

If i change the Serial1.begin(9600) to Serial1.begin(74880) I see that the ESP responds correctly to the AT command. It then gets an AT+RST command and indeed resets itself and I see the same message appear as given above. After the reset, communication starts to get problematic again, probably because the ESP switches to 9600 while the k210 does not. The sketch says that no WiFi shield is present.

btx000, 2019-04-12:
The correct baud rate supported by the esp8285 is 115200. 
I modified the examples and you can test it again.

GitJer, 2019-04-12:
It works!!!!
Thank you!!!!

closed 2019-04-13 by btx000

sasilva1998, 2020-04-12:
Hi, I am now trying to use ESP8285 to connect to a wifi network and getting that same error:
`OSError: couldn't init nic esp8285 ,try again please`
already checked baudrate  and 8 pin to be high, I am using the following firmware:
`MicroPython v0.5.0-42-g458ed4e on 2020-04-10; Sipeed_M1 with kendryte-k210`, will try with another build just in case.

Anyway, could there be another solution?

Seni0r, 2021-03-05:
> Try using logic analyzer to read the serial output when esp8285 first startup.

hi, the page is not found i dont know why

Seni0r, 2021-03-05:
> It works!!!!
> Thank you!!!!

can you send me the code please?
the page not found