sipeed/Maixduino #3: No OSX version yet?

andriyadi, 2019-04-19:
I did try to install on OSX, but the installer said: 
```Tool kflash is not available for your operating system.```

Please make it available.

btx000, 2019-04-20:
We updated the V0.3.7 version two days ago and have added the osx toolchain. But I haven't tested it on OSX yet, so I'm not sure if it works.

The board manager address below has been updated.
http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json

andriyadi, 2019-04-21:
I made a post of the workaround before 0.3.7 here: [https://www.hackster.io/andri/aiot-sipeed-maix-board-arduino-on-osx-a8455d](https://www.hackster.io/andri/aiot-sipeed-maix-board-arduino-on-osx-a8455d)

Possibly kflash.py will failed due to: `os.get_terminal_size()` as explained on my writing above.

closed 2019-04-27 by btx000