sipeed/Maixduino #8: how is this model generated?

laurentopia, 2019-05-16:
https://github.com/sipeed/Maixduino/blob/master/libraries/Maix_Speech_Recognition/examples/voice_control_led_en/voice_model.h

btx000, 2019-05-17:
Hi, @laurentopia 
We provide a simple routine for outputting the sound model. You only need to upload the program to the maix go board, speak to the board, and then the voice model will be printed on the serial port.
https://github.com/sipeed/Maixduino/blob/master/libraries/Maix_Speech_Recognition/examples/get_voice_model/get_voice_model.ino

Alternatively, you can use `rec.record(keyword_num, model_num)` in the program to save the model directly into memory, thus real-time adding models and recognition.

It should be noted that you can only set up to 4 voice models for a keyword, up to 20 keywords.

laurentopia, 2019-05-17:
Thank you @btx000.

closed 2019-06-07 by btx000