sipeed/Maixduino #12: Path Issue? (No such file or directory)

WaskaLabs, 2019-05-29:
Maixduino 0.3.8 and 0.3.7 tested on Arduino 1.8.8 and 1.8.9. The directory contains the files but for some reason are not being seen by arduino.

-- \Maixduino\hardware\k210\0.3.8/cores/arduino/kendryte-standalone-sdk/lib/bsp/include: No such file or directory

(happening with all files in cores/arduino/...)

Possibly an issue on my end, but not sure.

WaskaLabs, 2019-05-30:
UPDATE: The issue seems to be that my computers name contains a space. e.g. "name name2" ... Only name2 appeared on the path. I had to install it on another computer to figure it out. I have never had this issue with any other packages, so I assume its an easy fix.

Also, I was unaware that kflash required python3 and pyserial. May add that into the install docs. Otherwise it seems to work great!

Neutree, 2019-05-30:
Did you use Windows? python3 is not required!

WaskaLabs, 2019-05-31:
I used a windows first, then got it to work on a mac. kflash requires python3 and pyserial, I got compilation errors in the IDE (in verbose mode) that prompted me to pip install pyserial

btx000, 2019-07-05:
This problem is caused by a space in the username. this bug has been fixed in v0.3.9.

closed 2019-07-05 by btx000