sipeed/Maixduino #16: Unable to install examples from Arduino IDE

technoblogy, 2019-06-14:
I'm trying to get started with my Maixduino Kit by running some examples from the Arduino IDE, such as Blink, but I'm stuck. When I try and upload I get the error:

    Cannot run program "/usr/local/bin/python3": error=2, No such file or directory
    ...
    An error occurred while uploading the sketch

I'm using Arduino IDE 1.8.9 on a MacBook Pro. I've installed the Maixduino(k210) board version 0.3.8, and I've selected the "Sipeed Maix Go Board" option.

btx000, 2019-06-22:
You need to install python 3 and pyserial. See https://github.com/kendryte/kflash.py/blob/master/README.rst#macos-requirements

technoblogy, 2019-06-22:
I'm not keen on installing Homebrew. Is there any other way?

technoblogy, 2019-06-23:
I've installed python3 and pyserial. Now when I try and upload a simple program such as Blink I get:

    [WARN] Timeout Error, retrying... 
    [ERROR] Failed to Connect to K210's Stub 

or if I select the other port:

    [ERROR] No vaild Kendryte K210 found in Auto Detect, Check Your Connection or Specify One by`-p /dev/ttyUSB0`

btx000, 2019-07-05:
The arduino board was supported after v0.3.9. You can now update to resolve this issue.

closed 2019-07-11 by btx000

technoblogy, 2019-07-11:
Thanks - I confirm that I can now get the Arduino example programs to upload and run.