sipeed/Maixduino #18: Maixduino board unable to finalize the upload procedure of code

mrayy, 2019-06-17:
I just received Maixduino board from Seeedstudio:
https://www.seeedstudio.com/Sipeed-Maixduino-for-RISC-V-AI-IoT-p-4046.html

I understand version 0.3.9 should support this board. I set the board type to: Sipeed Maixduino Board in Arduino IDE.

I tried to compile and upload the code to the board, however after the "Downloading ISP" is finished, I get the following error:
"[WARN] Index Error, retrying... "

After several trials, the uploading procedure fails with the error:
"[ERROR] Failed to Connect to K210's Stub "

Please help how to resolve this issue...

btx000, 2019-06-26:
See here.
https://github.com/sipeed/Maixduino/issues/19#issuecomment-505797383

btx000, 2019-07-05:
Already updated to v0.3.9, if there is still a problem, please reopen this issue.

closed 2019-07-05 by btx000

jeanantoine, 2019-07-06:
> Already updated to v0.3.9, if there is still a problem, please reopen this issue.
Can you update **Platformio** please:
http://dl.sipeed.com/MAIX/platformio/dl-packages/manifest.json   04-May-2019 05:58:
` "framework-maixduino":[
        {
            "sha1": "e0bae63abe1b9071e59349460d5dfac5b42c7d26",
            "system": "*",
            "url": "http://dl.sipeed.com/MAIX/platformio/dl-packages/framework-maixduino-0.3.8.tar.gz",
            **"version": "0.3.8"**
        }
    ],
.........`

http://dl.sipeed.com/MAIX/platformio/dl-packages/framework-maixduino-0.3.9.tar.gz  05-Jul-2019 12:41   
http://dl.sipeed.com/MAIX/platformio/dl-packages/tool-kflash-kendryte210-v0.9.1.tar.gz   05-Jul-2019 12:35