sipeed/Maixduino #19: Can't install release v0.3.9

iZhangHui, 2019-06-26:
Arduino IDE 1.8.9 on Windows 10 couldn't display and install v0.3.9 release. So I can't use Maixduino Board.
[image]

iZhangHui, 2019-06-26:
@btx000 Could you help to check this issue?  Can't waiting for testing my Maixduino Board.

btx000, 2019-06-26:
Sorry, I don't have much time to deal with these issues right now. There are temporary solutions here.
* Replace kflash with the following program
[kflash_py.zip](https://github.com/sipeed/Maixduino/files/3329293/kflash_py.zip)
* Complete the clone from this repository to the core folder.

We will update the 0.3.9 version of the installation link within a few days.

jeanantoine, 2019-07-01:
Hi, I have the same problem with Ubuntu 18.04.
There is only version 0.3.8 in Arduino IDE board manager on Linux too.
The zip file is for Windows (kflash_py.exe), is there a version for Linux ?
thank you!

jeanantoine, 2019-07-02:
I have flashed with kflash_gui, it works !
[kflash_gui](https://github.com/sipeed/kflash_gui/archive/master.zip)

iZhangHui, 2019-07-04:
A week has passed. :(

btx000, 2019-07-05:
Now you can install v0.3.9 normally in arduino ide and pio ide. If you have any problems, please give me feedback.

lights0123, 2019-07-05:
@btx000 On PIO, I get the following error ([the same happens on Windows](https://github.com/sipeed/platform-kendryte210/issues/11)):
```
PackageManager: Installing framework-maixduino @ ~0.3.9
Error: Could not find a version that satisfies the requirement '~0.3.9' for your system 'linux_x86_64'
```

lights0123, 2019-07-05:
The above was fixed with [this issue](https://github.com/sipeed/platform-kendryte210/issues/12).

closed 2019-07-11 by btx000