sipeed/Maixduino #23: hardware timer

mzy2240, 2019-07-11:
Do you mind to provide a high level hardware timer? Like the timer for the Teensy board and the zerotimer for the SAMD21/51 board, with callback and interrupt. If a lot of efforts are needed, an example of some work around is greatly appreciated as well. Thanks!

btx000, 2019-07-11:
We have provided the Ticker library, which is based on hardware timers. Can it meet your requirements? Here is an example https://github.com/sipeed/Maixduino/blob/master/libraries/Ticker/examples/Blinker/Blinker.ino

mzy2240, 2019-07-11:
Thanks! Looks like it is the one I need. Does it support microsecond interval btw?

mzy2240, 2019-07-11:
I just checked the source code and it seems like the microsecond interval is not supported, do you mind to add that function later? Thank you very much!

btx000, 2019-07-11:
Done.

mzy2240, 2019-07-11:
Just test it and it works! Thank you so much!

closed 2019-07-11 by mzy2240