sipeed/Maixduino #32: WiFi API Compatibility Issues

iZhangHui, 2019-07-24:
Arduino official boards use almost the same wifi chip with Maixduino. In order to keep WIFI API compatibility, please port the official codes into Maixduino repo.

Arduino WIFI boards:
[ARDUINO UNO WIFI REV2](https://store.arduino.cc/usa/arduino-uno-wifi-rev2)
[ARDUINO NANO 33 IOT](https://store.arduino.cc/usa/nano-33-iot)

WIFI Library and firmware:
[WiFiNINA](https://github.com/arduino-libraries/WiFiNINA)
[Firmware for u-blox NINA W102 WiFi/BT module](https://github.com/arduino/nina-fw)

u-blox NINA:
[NINA-W1 series.pdf](https://github.com/sipeed/Maixduino/files/3424669/NINA-W1.series.pdf)

chegewara, 2019-07-24:
I asked this question on bbs and ive been told that nina-fw is used to handle wifi:
https://bbs.sipeed.com/t/topic/1007

iZhangHui, 2019-07-24:
@chegewara But it seems that the library and examples Maixduino supplied is not working.
Maixduino host CPU can't commicate with esp32 WIFI module.

Pali775, 2020-03-22:
I assume, that Maixduino WiFi connection still not fixed...
I need to conclude that buying Maixduino as IoT quick prototyping solution was a really big mistake... no support, no help, no improvements... It is a pitty...

marcusobrien, 2022-10-11:
@chegewara - I also can not get the EPS32 wifi module working with the c/C++ arduino or platformio frameworks - which firmware cna be used to get the ESP 32 Wifif working for C/C++ Maixduino aplications (not MaixPy  the wifi works for python apps).