sipeed/Maixduino #37: "mobile net init fail" on MAIXduino

KotNitro, 2019-07-30:
Hello!

I ran into a bug with "mobilenet_v1" example on my MAIXduino - it begins to load model file, but "mobile net init fail" appears on serial port monitor.
I tried to find the reason and I saw `mbnet.begin(kmodel_name)` return value -6.
I used different MicroSD cards and model files from your site with the same result. As `KPUClass` source code is not available (I did not find it), I can't debug it any further.
Please help to solve that.

Many thanks!
With regards,
Maksim

chegewara, 2019-07-30:
What maixduino version are you using?
There was issue that has been fixed https://github.com/sipeed/Maixduino/commit/240bb27d0b53c2ddb246a8bdf2a10692bdc001dc
and is merged to v0.3.10

KotNitro, 2019-07-31:
Version 0.3.10 is shown in list in Arduino IDE, but is not actually available (previous versions still could be downloaded):
> Error downloading http://dl.sipeed.com/MAIX/Maixduino/core/Maixduino_core_v0.3.10.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading http://dl.sipeed.com/MAIX/Maixduino/core/Maixduino_core_v0.3.10.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Возникла ошибка при загрузке http://dl.sipeed.com/MAIX/Maixduino/core/Maixduino_core_v0.3.10.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

chegewara, 2019-07-31:
First of all i can download v0.3.10 without problem, maybe try one more time, and second of all i think your problem is duplicate of this:
https://github.com/sipeed/Maixduino/issues/29

KotNitro, 2019-07-31:
I have updated MAIXDuino to V0.3.10, but I still see those messages on port monitor:
> init mobile net, load kmodel from SD card, it may takes a long time

and after about 10 seconds
> mobile net init fail

So, it looks like loading has been started, but crashes a bit later for some reason.
I use `mobilenet_0x300000.kfpkg` file renamed to `m` according to code. I also see "Loading..." on LCD, so file was found on uSD and it is not well-known uSD bug.

chegewara, 2019-07-31:
Maybe try different card since it is "Loading...", which means card has been detected and file possibly has been found, i didnt try code with option: insert card with no file on it (i am using name 'model').

KotNitro, 2019-08-02:
I tested a number of cards, 16 and 32 Gb, with the same result. 
Are there any limitations for card size and file system?

KotNitro, 2019-08-03:
Now it works!
In my case it was important to satisfy some requirements:
- MAIXduino version is 0.3.10 (the latest for today)
- MicroSD size is 16 Gb (8 Gb does not work for some reason) and formatted with FAT32 file system
- downloaded mobilenet_0x300000.kfpkg **needs to be unpacked** as it is actually .zip file!
- unpacked model file name should be the same as in code (single letter `m` without any extension by default).
Thanks for help

closed 2019-08-03 by KotNitro

excalibur-sa, 2019-12-18:
I changed a 16G size tf card to resolve it