sipeed/Maixduino #46: 400 / 600 mhz no difference in performance

dimitre, 2019-08-07:
Hello, I've just tested some benchmark at 400Mhz and 600Mhz and the performance outputs the same. I think this options aren't working on Maixduino yet.
https://github.com/PaulStoffregen/CoreMark
Thanks
[image]
[image]

CombiesGit, 2019-12-15:
`void setup() 
{
  sysctl_cpu_set_freq(F_CPU);
}

void loop() 
{

}`

Are there any side effects?
I dont know.

romainreignier, 2022-10-11:
Indeed, I have looked into the source code and the menu entry does not attempt to change the CPU frequency anywhere.
It only changes the `SystemCoreClock` constant but it is not used anywhere, not even in the `millis()` implementation.