sipeed/Maixduino #48: K210 not detected

baqwas, 2019-09-03:
Hello!
I have used two different cables and two different ports to connect to Maixduino from Ubuntu 19.04 desktop. I have had ZERO luck. Attached is a summary of my exercise. I can provide additional details upon request.

Will someone at Sipeed be kind enough to help me with this persistent and repeatable issue?

Xie xie.
[MicroPython local build firmware.docx](https://github.com/sipeed/Maixduino/files/3570837/MicroPython.local.build.firmware.docx)

vanvuongngo, 2019-10-15:
Hi baqwas,

I am using ubuntu 19.04 as well. Did you executed following to have access to the serial port?
```
sudo usermod -a -G dialout $(whoami)
```
see install documentation under https://maixduino.sipeed.com/en/get_started/install.html
I use platform.io instead of Arduino IDE but the preparation instruction is the same.

baqwas, 2019-10-15:
Hello Van,

Thank you for your suggestion. Of course, I did set the permission (along the lines _you have suggested_) a very long time ago! I have five other Arduino boards (different types) working without any issue. I can upload to these boards from the Arduino default IDE without any issue (except for the occasional reset for operational reasons). Yes, I am diligently ensuring that the right settings combinations board/port/flash are set before attempting to upload to a specific board.

I understand the productivity advantages of **platform.io** especially when it is used in conjunction with, for example, Visual Studio Code but my needs are very modest. Right now I want to keep things simple and just focus on reading/writing to one or two pins let alone work on RISC-V magic that may happen later.

FWIW, @loboris provided an alternate development, **MicroPython**, approach that I found very interesting.

Kind regards.

vanvuongngo, 2019-10-15:
Hi Matha,

I can reproduce your problem and it could be that the automatic recognition of the board does not work for you either. You can try to set a board manually with the argument ```-B```. Possible options are kd233, dan, bit, goD, goE or trainer. I have a MaixGo and it works when I execute following (port automaticlly detected on ```/dev/ttyUSB1```):
```
./kflash.py -BgoE -b 2000000 -t MaixPy.bin
```
Hope it will helps you and I can understand if you want to use ```python```. It's not my favorite programming language but it also has its advantages.

vanvuongngo, 2019-10-15:
Or use ```MicroPython``` from sipeed instead, see https://maixpy.sipeed.com/en/get_started/upgrade_firmware.html

1. get a firmware (file with ```.bin``` extension) from https://github.com/sipeed/MaixPy/releases
2. get ```kflash_gui``` flasher from https://github.com/sipeed/kflash_gui
3. install ```MaixPy IDE``` link can be found under http://dl.sipeed.com/MAIX/MaixPy/ide/ or use a serial port command line tool like ```screen```

vanvuongngo, 2019-10-15:
@baqwas, your problem would be better placed in the MaixPy area see https://github.com/sipeed/MaixPy

baqwas, 2019-10-16:
Hello Van,

You are correct wtih your observation about the -**BgoE** option. That was the only way I could get the MicroPython to flash.

I'll try your suggestion on the MaixPy IDE. Right that is probably the more productive option for me. Again, really appreciate your assistance and guidance. Without your feedback (and that from @loboris), I would be floundering.

Kind regards.

P.S. The Serial-USB-C chip has apparent speed limitations well below the published spec;  I have to use a lower speed ~1.1Mbps. Perhaps, this is the root cause of my issue? I am plugging directly from the (ASUS H370-PRIME) motherboard to the board (and I have tried three different cables) - no external hub.

vanvuongngo, 2019-10-16:
> P.S. The Serial-USB-C chip has apparent speed limitations well below the published spec; I have to use a lower speed ~1.1Mbps. Perhaps, this is the root cause of my issue? I am plugging directly [...]

Nice and I guess the ```kflash.py``` script has a bug but I did not debug it. Feel free to close this issue when you can work now.

matheus-giordani, 2024-02-21:
I have the same problem but in Windows 10. I already installed the drivers following the step by step instructions in the documentation [tutorial](https://wiki.sipeed.com/soft/maixpy/en/get_started/install_driver/duino.html), but so far nothing, could anyone help me?
board: k210