sipeed/Maixduino #49: Cannot download support library

baqwas, 2019-09-03:
Hello!
I cannot download the libraries for the Maixduino board under Arduino IDE. Two screen shots are attached for reference purposes.

Can anyone help me to resolve the issue, please?

Thanks.
[image]
[image]
[image]

chegewara, 2019-09-03:
There is 2 json files to download maixduino, please try the other one:
http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json

baqwas, 2019-09-03:
Hello @chegewara 

I've attached the screen shots of the Board Manager settings in use. I do believe that the settings are the default ones for the **Maixduino** board with **USB0** and **k-flash**. Appreciate the quick response but I'm still stuck.

Kind regards.
[image]
[image]
[image]

chegewara, 2019-09-03:
From screenshots it looks like you have all installed and yes, k-flash is the correct one programmer.
The most recent version is 0.3.10 and i dont have problems to flash maixduino on my ubuntu 14.10 and arduino-ide.

Just in case you can try to turn on flashing mode with onboard buttons (press both btns and 1st release pwr).

baqwas, 2019-09-04:
@chegewara 

> turn on flashing mode with onboard buttons (press both btns and 1st release pwr)

That is precisely what I have done _repeatedly_ prior to seeking help by logging an issue! That is my routine operational mode with Arduino when there there are unpredictable freezes in the output pane.

The reason I'm pursuing this issue after a lapse of attention is that the Maixduino and the Grove AI HAT both have similar issues (my simplistic assumption) - **not detecting K210**. It is quite conceivable that the two anomalies are distinctly separate but the inability to run the simplest standard exercise using all the documented steps (except that I'm on Ubuntu 19.04) for both boards is perplexing owing to my limited knowledge in these matters.

Kind regards.

chegewara, 2019-09-04:
Yes, it can be the reason on Ubuntu 19.04.

baqwas, 2019-09-04:
In that case, let me wait for Ubuntu 19.10 next month and assess whether that would help out.

Thanks everyone.

closed 2019-09-04 by baqwas

yulonglovecode, 2020-05-21:
hi i still face this problem , anyone know why i cant download the library with the URL given for the preference given?