sipeed/Maixduino #51: Lower Baud Rate

baqwas, 2019-09-05:
Hello!

For the **Maixduino** board, the Arduino IDE (v1.8.9) currently provides only 1.0 & 1.5 Mbps support for code upload. @loboris identified that the CH552T has an [issue](https://github.com/diodep/ch55x_dualserial) that inhibits its use at these settings. I could flash his MicroPython firmware only at this speed. The higher values (viz. >= 1.0) failed to detect the MCU to initiate the code upload.

With the Arduino IDE, we are accustomed to using 0.1152 Mbps for serial communications. Is it possible to provide such support with the K210 library that you have published? Your considerate attention to this request would help us to make better use of Maixduino in our applications.

chegewara, 2019-09-05:
That strange, because its working for me with 1.5Mb without any issues.
You can change this file to add 115200 and test it on your setup:
https://github.com/sipeed/Maixduino/blob/master/boards.txt#L70-L76

baqwas, 2019-09-05:
@chegewara 

Thanks a bunch. I will definitely give it a try after a cursory review (not that I understand exactly what is going on beneath the covers ;)) but simply to improve my learning. I will let you know as soon as practical on the results. Right now, my tactical focus is to ensure that I have a basic understanding of the available library of MicroPython modules (because my skimpy Python knowledge is interfering). Again, appreciate your continuing support.

FWIW, I suspect that @loboris' suggestion to use the `-B goE` option and the lower baud rate must have been _the combo to unlock my issue_. As you mention, I did try 115200 bps, 0.75, 1 & 1.5 Mbps earlier and was not successful. Now that 1152000 will let me flash I do not want to press my luck with a faster speed right now. As you can understand, MCU hardware is not my expertise.

Of course, I have to manually RST the board after a flash although the schematic and specs indicate (per @loboris) that it should have been automatic.

Kind regards.

P.S. I've been waiting for help since the end of May. I am very fortunate that @loboris and you pushed me in the right direction. Many thanks again.

closed 2019-09-05 by baqwas