sipeed/Maixduino #54: FreeRTOS in Maixduino

maxpromer, 2019-09-22:
Hi, i see you use kendryte-standalone-sdk so it not support FreeRTOS to create task but want to use FreeRTOS because my application needs do everything in same time. so i think i will implement FreeRTOS by self. but i have no idea about implement FreeRTOS.

Can you  recomment something for me ?

chegewara, 2019-09-22:
https://github.com/kendryte/kendryte-freertos-sdk

Neutree, 2019-09-23:
kendryte removed FreeRTOS from master branch, and you can use freertos branch: https://github.com/kendryte/kendryte-standalone-sdk/tree/freertos

maxpromer, 2019-09-23:
> kendryte removed FreeRTOS from master branch, and you can use freertos branch: https://github.com/kendryte/kendryte-standalone-sdk/tree/freertos

I clone branch this to `/cores/arduino/kendryte-standalone-sdk` and i upload this code to Sipeed Maix One Dock but not work. new task not do.

```c++
#include <FreeRTOS.h>
#include <task.h>

void vTaskCode( void * pvParameters ) {
  for ( ;; ) {
    Serial.println("Run in task !");
    vTaskDelay(500 / portTICK_PERIOD_MS);
  }
}

TaskHandle_t xHandle = NULL;

void setup() {
  Serial.begin(115200);

  Serial.println("Task create");
  xTaskCreate(
    vTaskCode,       /* Function that implements the task. */
    "vTaskCode",     /* Text name for the task. */
    1024,            /* Stack size in words, not bytes. */
    NULL,            /* Parameter passed into the task. */
    4,               /* Priority at which the task is created. */
    &xHandle );      /* Used to pass out the created task's handle. */
}

void loop() {
  Serial.println("Main loop");
  vTaskDelay(500 / portTICK_PERIOD_MS);
}
```

[image]

What i do wrong ?

Neutree, 2019-09-24:
@maxpromer 

Only one task was created, and scheduler not run yet in your code.

You should add some code to loop function:
```
vTaskStartScheduler();
while(1){}
```

the maixduino not run FreeRTOS scheduler by default, so you should run scheduler in your  loop

You will understand all once you saw the source code as follow:

https://github.com/sipeed/Maixduino/blob/b10b99a10bff9796061fb373ac3ec7249d436be5/cores/arduino/main.cpp#L24-L28

maxpromer, 2019-09-24:
> @maxpromer
>
> Only one task was created, and scheduler not run yet in your code.
>
> You should add some code to loop function:
>
> ```
> vTaskStartScheduler();
> while(1){}
> ```
>
> the maixduino not run FreeRTOS [...]

It work ! Thank so much. freertos branch i don't see example to use FreeRTOS so i have not a way to understand use FreeRTOS (I'm not professional about FreeRTOS. i'm user)

And this full code of work.

```c++
#include <FreeRTOS.h>
#include <task.h>

void vTaskCode( void * pvParameters ) {
  for ( ;; ) {
    Serial.println("Run in task !");
    vTaskDelay(500 / portTICK_PERIOD_MS);
  }
}

void vTaskCode2( void * pvParameters ) {
  for ( ;; ) {
    Serial.println("Run in task 2 !");
    vTaskDelay(500 / portTICK_PERIOD_MS);
  }
}

TaskHandle_t xHandle = NULL, xHandle2 = NULL;

void setup() {
  Serial.begin(115200);

  Serial.println("Task create");
  xTaskCreate(
    vTaskCode,       /* Function that implements the task. */
    "vTaskCode",     /* Text name for the task. */
    1024,            /* Stack size in words, not bytes. */
    NULL,            /* Parameter passed into the task. */
    4,               /* Priority at which the task is created. */
    &xHandle );      /* Used to pass out the created task's handle. */

  xTaskCreate(
    vTaskCode2,       /* Function that implements the task. */
    "vTaskCode2",     /* Text name for the task. */
    1024,            /* Stack size in words, not bytes. */
    NULL,            /* Parameter passed into the task. */
    4,               /* Priority at which the task is created. */
    &xHandle2 );      /* Used to pass out the created task's handle. */
}

void loop() {
  vTaskStartScheduler();
  while(1){}
}
```

closed 2019-09-24 by maxpromer

maxpromer, 2019-09-24:
I test code on Maixduino v0.3.10 the FreeRTOS work without clone freertos branch replace to `/cores/arduino/kendryte-standalone-sdk` so you can use FreeRTOS right now.