sipeed/Maixduino #59: cannot declare variable 'camera' to be of abstract type 'Sipeed_OV2640'

sabas1080, 2019-10-14:
Hi

I have the next error `cannot declare variable 'camera' to be of abstract type 'Sipeed_OV2640'`

Arduino IDE version 1.8.9 with example mobilenet and selfie

Thanks

tjmc, 2019-10-17:
setRotation is misspelled. If you change setRotaion to setRotation it'll work.

https://github.com/sipeed/Maixduino/blob/85c0d7d24e652004c3a7ef8bec3ca719129db1b3/libraries/Sipeed_OV2640/src/Sipeed_OV2640.cpp#L724

https://github.com/sipeed/Maixduino/blob/85c0d7d24e652004c3a7ef8bec3ca719129db1b3/libraries/Sipeed_OV2640/src/Sipeed_OV2640.h#L57

closed 2019-10-18 by sabas1080

garyinbg, 2019-11-22:
I see this is closed however it is very much open for me :)
I have exactly the same error popping up and have obviously chacked the above, downloaded the most recent versions in which the typo was corrected and tried to compile ony be be met with the same error...
Is there something else available to get me past this please before I start the line by line operations 🗡 
Many thanks in anticipation  Sabas, did you get this running ?

pernicious-flier, 2019-11-22:
same for me as garyinbg

pernicious-flier, 2019-11-22:
garyinbg , i had to fix those two errors in C:\Users\xxxxx\AppData\Local\Arduino15\packages\Maixduino\hardware\k210\0.3.11\libraries\Sipeed_OV2640\src directory and now works..

technoblogy, 2020-03-17:
This should not be closed until the two typos have been fixed in the library. They are still there as of March 2020 (version 0.3.11).

nitz08, 2020-03-30:
I fixed those errors but i am getting snap fail on the serial monitor and black screen on LCD. was anyone successful to implement that example code?

technoblogy, 2020-03-31:
Yes, the example program:

    Examples->Sipeed_OV2640->selfie

worked for me once I made the fixes.

bandekarsm, 2020-05-29:
worked for me too.

paipeng, 2020-05-29:
after fixing the misspelling, I can't open the camera (begin() returns false). any suggestions?

pepperm2, 2020-06-06:
I have the same failure with the "selfie.ino" example. I have 0.3.11 on Arduino 1.8.12 and just get "snap fail" in the serial window and a red display screen.

Has anyone got any more ideas as I have just started experimenting with this board in lieu of the article series starting to appear in Elektor magazine.

paipeng, 2020-06-13:
Hello @pepperm2 ，

I have  heard from developer that the new board shipping with another camera module, not OV2640. You need to attach a OV2640 camera module on the board and the code works now. I have just tested it.

Regards,

Pai

pepperm2, 2020-06-13:
I’ve cracked it...I think, for me anyway...

Apart from the spelling mistakes mentioned previously in the code, the camera will plug into the connector two ways around. When you put it in the wrong way around the camera is destroyed (because of the inevitable reverse polarity supply I guess) and won’t then work when put in the correct way around. The cable is marked 01 to 24 and the connector marked “1” on the PCB. In hind sight, the “01” and “1” should be aligned :-) I purchased a new OV2640 for about £5 on eBay, fitted it the correct way around, and now everything works camera wise on my MaixDuino.

I wonder if other people have had this problem?

oliverjc, 2020-06-14:
Alrighty, I had many troubles in trying to get the mobilenet_v1 example to work and ended up here. I'm sure other people will see this when trying to get the Maixduino working so here we go...

1. If you're using anaconda you need to set the path for your python installation at `/Users/<user>/Library/Arduino15/packages/Maixduino/hardware/k210/0.3.11/platform.txt` to `/usr/local/anaconda3/bin/python3` (just CTRL-F "python" in that file and you'll see what i mean)
2. Change `setRotaion` to `setRotation` in both the .cpp and .h file in `/Users/<user>/Library/Arduino15/packages/Maixduino/hardware/k210/0.3.11/libraries/Sipeed_OV2640/src` (come on guys.... fix this - this is definitely not a closed issue!)
3. The camera that ships with the Maixduino doesn't seem to be an OV2640. Fortunately, I had another laying around and it worked when I swapped it in. Pins should be facing **UPWARDS** when inserted. 
4. You will be very disappointed by the results of all these efforts once it starts working - the mobilenet_v1 example is just terrible.

user-emx, 2020-06-17:
I have MaixDuino (2626), and examples in python language work, especially the camera works.
However, this is not the case with the arduino-ide example in cpp: selfie.ino. Procedure
int Sipeed_OV2640::sensor_snapshot( ){    g_dvp_finish_flag = 0;
    uint32_t start =  millis();  while (g_dvp_finish_flag == 0)    {
        usleep(50);  if(millis() - start > 300)  return -1; }
    return reverse_u32pixel((uint32_t*)_dataBuffer, _width*_height/2); 
} returns minus one, which is why image pointer *img of selfie.ino is NULL.
As the internet shows, i am not the only one with this problem.

user-emx, 2020-06-19:
My Sipeed Maixduino (2626) kit is with a GC0328-camera. 
The flexible flat ribbon cable of 2cm has the inscribtion: zv-t01-ga4.4.
The solution of my problem was very close to me in the form of an ESP32 kit with a OV2640-camera.
Now selfie.ino (with PlatformIO selfie.cpp) works as expected.
Sipeed_OV2640.cpp scans for the GC0328-camera, but it doesn't work that way of code.

dimitre, 2020-08-09:
Actually having issues with camera too. this should not be closed.

AndroGiorgi, 2020-11-05:
My camera work with MaixPy, but i view blank red display with Arduino and snap fail message in serial monitor.

Cwilson545, 2022-01-05:
Hey guys,

I've recently attempted to run the 'Selfie' sketch after making the adjustments to the two files listed above. I seem to get an error "Greeting fail, check serial port ([ERROR] No vaild Kendryte K210 found in Auto Detect, Check Your Connection or Specify One by`-p COM3`  )". This board is brand new from the seeedstudio.com website
[image]
, so any ideas on how to fix this?

ITstreet1, 2023-10-24:
Just to confirm, it is the end of 2023, and those files are still without a fix. You still have to change Rotaion to Rotation.
There are problems with adafruit gfx lib, as it must be modified meanwhile

canDry, 2023-11-16:
> Just to confirm, it is the end of 2023, and those files are still without a fix. You still have to change Rotaion to Rotation. There are problems with adafruit gfx lib, as it must be modified meanwhile

Yah.  Sipeed has abandoned this product ages ago - like many Chinese boards they release them, then their immediately obsolete.  No reason to ever buy sipeed products!