sipeed/Maixduino #62: maixduino not detected in macOS Catalina

auvi, 2019-10-27:
Hello,
I am having a hard time making the Maixduino detected in macOS Catalina. I have used a USB-C to USB-C data cable to connect the Maixduino to my MacBook Pro 13" (2017). I also have the Maixduino powered up and it shows "Welcome to MaixPy" in the LCD screen, the reset button also works. But so far macOS doesn't detect the board. I have checked ls /dev/cu.* and nothing changes after I connect the board to the mac. Anybody successfully worked with the Maixduino board from macOS Catalina (version 10.15)?
Thanks

mat-lo, 2020-05-12:
I have the same issue. The board seems working as it reboots and I see the flashing LEDs, but I can't see it in my serial ports. Running OSX 10.15.4.

mat-lo, 2020-05-12:
Solved - The issue is that the Maixduino board doesn't have a pull-up resistor on CC, so you can't power it with a USB-C to USB-C cable.  Use a USB-A to USB-C cable, with an adapter and it will work. Also don't forget to install the FTDI drivers.