sipeed/Maixduino #69: M1&M1w dock Vs Maixduino Vs SpeechRecognizer

revgagne, 2019-12-09:
Because the pins number are directly specified instead of using the constant defined in :

/Maixduino/hardware/k210/0.3.11/variants/sipeed_maix_one_dock/pins_arduino.h

The SpeechRecognizer and it's example are unusable on Dan Dock AKA M1&M1w and possibly other Maix products.

The simple solution is to simply specify SpeechRecognizer::begin() like this:

int
SpeechRecognizer::begin()

    //io_mux_init
    fpioa_set_function(MIC0_DATA, FUNC_I2S0_IN_D0);
    fpioa_set_function(MIC0_BCK, FUNC_I2S0_SCLK);
    fpioa_set_function(MIC0_WS, FUNC_I2S0_WS);

---

xgarb, 2019-12-15:
Hi, Do you know how I can stop the speech recognition once I have what I need?

I'm doing a multi-factor auth system including a spoken pass phrase. I need something like rec.end() when this part is complete because otherwise the speech recognition continues to run. Code is something like this...

```
  rec.begin();
  rec.addVoiceModel(0, 0, red, fram_num_red);
  rec.addVoiceModel(1, 0, green, fram_num_green);
  rec.addVoiceModel(2, 0, blue, fram_num_blue);

  long test_start_millis = millis();
  int i = 0;
  while (millis() - test_start_millis < security_time_out) {
    Serial.println("voice waiting");
    int res;
    res = rec.recognize(); 
   ... etc
```

revgagne, 2019-12-16:
Use break to end your while loop before security_time_out if this is what you wish!...

revgagne, 2019-12-16:
Each time a phrase is recognize, spch_recg() is stopped!  You're restarting it inside your while by using res.recognize() again and again!

xgarb, 2019-12-17:
Maybe it's a problem elsewhere. With the code below the text on the LCD 'Before voice' is instant. The 'After voice' is slowly written across the screen.

```
#include <Sipeed_ST7789.h>
#include "Maix_Speech_Recognition.h"
#include "voice_model.h"

SpeechRecognizer rec;
SPIClass spi_(SPI0); 
Sipeed_ST7789 lcd(320, 240, spi_);

void auth_voice()
{
  rec.begin();
  rec.addVoiceModel(0, 0, red, fram_num_red);
  int res;
  res = rec.recognize();
  Serial.printf("res : %d ", res);
}

void setup() {
  Serial.begin(115200);
  lcd.begin(15000000, COLOR_BLACK);
  lcd.setTextSize(2);
  lcd.setCursor(10, 40);
  lcd.println("Before voice...");
  auth_voice();
  lcd.setCursor(10, 80);
  lcd.println("After voice...");
}

void loop() {
}
```
When the word 'red' is recognised I see this in Serial...

> speeking...
> vad ok
> mfcc ok
> no. 1, frm_num = 19, save_mask=12345cur_dis=417
> [INFO] recg cycle = 0x001a8acf
> res : 1

revgagne, 2019-12-24:
There seem to have a contingency between i2s for microphone and spi for the LCD display... Serial.print/Serial.println doesn't seem to be affected... Try to avoid using the LCD if your fidling with the i2s microphone interface...  And yes, this seem to be a bug somewhere!...