sipeed/Maixduino #73: snap fail

ouening, 2020-01-21:


biokys, 2020-01-28:
same problem here...cant continue

freetoair, 2020-02-11:
same problem, 
does anyone intend to respond ???

biokys, 2020-02-11:
when i run the same code in micropython, everything works like a charm. the similar code in arduino fails in getting a snapshot from camera

nitz08, 2020-03-28:
I am having the same issue. Did anyone figure it out?

lakidd, 2020-03-29:
And another one... works in uPython fine..

nitz08, 2020-03-31:
@lakidd The thing is I have to use C++/c for my project. I tried MaixPy IDE and I am getting "sensor timeout error". For the face detection example, after flashing face_model, it works properly. But the hello_world code(snap) I am getting error. 
Also, Platfrorm IDE did not work. 
Now, I am trying with Kendryte SDK. There I am having an issue with CMake files.  
Is this device that poorly supported?

nsch0e, 2020-04-01:
same problem here: using the standard camera with maixduino (gc0328) doesnt work with arduino example, but works in microPython.

omnivision ov2640 camera works with both sketches. So probably there is no initialization code for the gc0328 camera...

JonRobo, 2020-05-26:
Works in micropython for me as well but no luck in Maixduino...
I tried to increase the timeout limit in Sipeed_OV2640::sensor_snapshot(), but no luck. It seems to be dependent on dvp_get_interrupt(DVP_STS_FRAME_FINISH) to turn true. Digging into the dvp.h gets into the HAL which is a bit messy.

I am using the M1W Maix Dock. Maybe there is a pin define error? I hope there is a fix, I want to eventually run my custom Kmodel using mobilenet_v1 example

user-emx, 2020-06-16:
I have MaixDuino (2626), and examples in python language work, especially the camera works.
However, this is not the case with the arduino-ide example in cpp: selfie.ino. Procedure
int Sipeed_OV2640::sensor_snapshot( ){    g_dvp_finish_flag = 0;
    uint32_t start =  millis();  while (g_dvp_finish_flag == 0)    {
        usleep(50);  if(millis() - start > 300)  return -1; }
    return reverse_u32pixel((uint32_t*)_dataBuffer, _width*_height/2); 
} returns minus one, which is why image pointer *img of selfie.ino is NULL.

patrickpollet1966, 2020-06-18:
Dear All,

I have the same problem and noticed something interesting.

The snapshot works with MaixPy and not with arduino sketch.
But if you write the python code under Repl you may notice that sensor.reset() returns 'gc0328' which is a camera supported by MaixPy (the driver is included) but not by the arduino library for Maixduino.

I ordered an ov2640 to replace the gc0328 for a few bucks and i will see if it works with arduino sketch (i think so).

Regards

user-emx, 2020-06-19:
You are certainly right. I have GC0328 too, and for a HELLO_WORLD I changed Sipeed_OV2640::sensor_snapshot( ) to:
 int Sipeed_OV2640::sensor_snapshot( ) {	
    g_dvp_finish_flag = 0;
    //----------begin--------------
    uint16_t index = 0;   
    cambus_writeb(GC0328_ADDR, 0xfe, 0x01);
    for (index = 0; sensor_default_regs[index][0]; index++) {
        if(sensor_default_regs[index][0] == 0xff){
            usleep(sensor_default_regs[index][1]);
            continue;
        }
        cambus_writeb(GC0328_ADDR, sensor_default_regs[index][0], sensor_default_regs[index][1]);
    }
    dvp_start_convert();
    //----------end---------------
    uint32_t start =  millis();
    while (g_dvp_finish_flag == 0) {
        usleep(50);
        if(millis() - start > 300) return -1;
    }
    return reverse_u32pixel((uint32_t*)_dataBuffer, _width*_height/2);
}
and now the pointer is not NULL and I get fragments of an image.
You have to add    uint8_t sensor_default_regs[][2] = {{0xfe , 0x80},  …}    to Sipeed_OV2640.cpp

user-emx, 2020-06-19:
My Sipeed Maixduino (2626) kit is with a GC0328-camera. 
The flexible flat ribbon cable of 2cm has the inscribtion: zv-t01-ga4.4.
The solution of my problem (with help of PatrickPollet1966) was very close to me in the form of an ESP32 kit with a OV2640-camera.
Now selfie.ino (with PlatformIO selfie.cpp) works as expected.
Sipeed_OV2640.cpp scans for the GC0328-camera, but it doesn't work that way of code.

user-emx, 2020-06-23:
With the OV2640 you will have success. For me selfie.cpp works now.
Sincerely: Edgar

From: patrickpollet1966
Sent: Thursday, 18 June 2020 13:56
To: sipeed/Maixduino
Cc: user-emx; Comment
Subject: Re: [sipeed/Maixduino] camera snap fail (#73)

Dear All,
I have the same problem and noticed something interesting.
The snapshot works with MaixPy and not with arduino sketch.
But if you write the python code under Repl you may notice that sensor.reset() returns 'gc0328' which is a camera supported by MaixPy (the driver is included) but not by the arduino library for Maixduino.
I ordered an ov2640 to replace the gc0328 for a few bucks and i will see if it works with arduino sketch (i think so).
Regards

patrickpollet1966, 2020-06-23:
@user-emx, 

Yes i am sure to succeed.
The GC0328 have absolutely no chance to works with Arduino as there's no driver, but the ov2640 have one.

If i found time for that, i think to port the gc0328 driver from MaixPy to Arduino. It will allow to use this camera with Selfie.cpp.

The ov2640 just arrived yesterday and i did not do the test until now.

barni2000, 2020-07-26:
There is a library for gc0328. https://github.com/fukuen/Maixduino_GC0328

user-emx, 2020-07-28:
Tank you very much

Edgar Marx

From: Barnabás Czémán
Sent: Sunday, 26 July 2020 14:36
To: sipeed/Maixduino
Cc: user-emx; Mention
Subject: Re: [sipeed/Maixduino] camera snap fail (#73)

There is a library for gc0328. https://github.com/fukuen/Maixduino_GC0328

micky008, 2020-07-28:
Thank you barni2000

installation [for windows 10  sorry unix guys]:
download the master branch of the repo.
extract the contant of GC0328 in this folder C:\Users\<Your user>\AppData\Local\Arduino15\packages\Maixduino\hardware\k210\0.3.11\libraries

rename the Maixduino_GC0328-master to Maixduino_GC0328.

finaly we got this :
C:\Users\<Your user>\AppData\Local\Arduino15\packages\Maixduino\hardware\k210\0.3.11\libraries\Maixduino_GC0328

and finish go to the selfi exemple and replace all "Sipeed_OV2640" by  : Maixduino_GC0328

it's over !

rumoldus, 2020-08-08:
micky008,  barni2000, super. Worked for me!

dimitre, 2020-08-09:
I'm having the same issue here with Platformio CLI, platform K210 normal or master.
I can't find the file @micky008 pointed out for the workaround. I've seen this library here but can't make it work
https://github.com/fukuen/Maixduino_GC0328

rumoldus, 2020-08-10:
@dimitre 
Please be more clear. Obviously you could find the library. The solution Micky008 suggests (and that worked for me) is for the Arduino IDE and you are referring to Platformio CLI. Describe your environment, what you tried, what error(s) you get.
Thanks

dimitre, 2020-08-11:
ok @rumoldus sorry about the confusion. for short just wanted to report "camera snap fail" here too for the new camera. it is working with the previous one.

gouldpa, 2020-08-28:
I followed barni2000 and micky008 instruction and it works. Thanks

rumoldus, 2020-08-28:
Did any of you with GV0328 camera get the mobilenet_v1.ini demo app to work ? My Maix is always returning the first object of the list (tench, Tinca tinca). I got the selfie.ini to work as reported earlier.

gouldpa, 2020-08-29:
After many hours work, I could not get it working. I also had (tench, Tinca tinca) on the serial but nothing on the screen. It would always crash when I tried to write to the LCD. I have just purchased an ESP-CAM and used it's camera (ov2640) on the Maixduino. The standard Selfie works. Will try the mobilenet_v1.ini next with this camera.

gouldpa, 2020-08-29:
mobilenet_v1.ini is up and running with the ov2640 from the ESP-CAM. Only adding the missing "t" on the line Sipeed_OV2640::setRotation in the Sipeed_OV2640.h and .cpp files.

rumoldus, 2020-08-30:
gouldpa / Paul,
I was a little further with the GV0328 and the mobilenet_v1 demo. I have display on the LCD, including the picture from the GV0328 camera, on the display. But the detection always gives 'tench, Tinca tinca'. It is frustrating.

TA-Robot, 2021-01-19:
In my case, using GC0328 with mobilenet_v1 demo got core dump.
It looks like mbnet._count always returned 4000. Is this normal?
Anyway, I want a perfect version of sample code which using GC0328 as mobilenet_v1 camera.

Vossen98, 2021-08-10:
> Thank you barni2000
>
> installation [for windows 10 sorry unix guys]:
> download the master branch of the repo.
> extract the contant of GC0328 in this folder C:\Users<Your [...]

Hello, 

I also downloaded the library and changed "Sipeed_OV2640" by : Maixduino_GC0328. In my program, the GC0328 has to take a picture but it has to be saved on an sd-card. The picture is saved as a .BMP on the sd-card but when I open the picture, it says that it doesn't support the file format. Does somebody know why?

Regards

dimitre, 2021-08-10:
I would try to open the final file in an hex editor to check the magic bytes and see if it matches BMP file.

abhi-84, 2021-11-25:
> Thank you barni2000
>
> installation [for windows 10 sorry unix guys]: download the master branch of the repo. extract the contant of GC0328 in this folder C:\Users<Your [...]

This works for unix/linux systems for arduino code with GC0328 camera. This camera module have ZV-T01-GA4.4 mentioned on the camera connector strip (in case of confusion with camera module no.)
The same procedure works for unix/linux guys, just go to the following path and follow as mentioned by @micky008 :
/home/xxxx/.arduino15/packages/Maixduino/hardware/k210/0.3.11/libraries

xxxx--> your username

Thanks for this fix