sipeed/Maixduino #76: ESP32 can't communicate

zpowellman, 2020-03-05:
ESP32 doesn't respond to either USB com or commands from K210 uP.  What firmware is on the ESP32?  I've verified AT commands are reaching the ESP32 but it doesn't respond.  Writing to Serial1  using any appropriate example shows no indication on the ESP LEDs.

technoblogy, 2020-03-17:
I confirm this problem with a Maixduino. Running the ScanNetworks example just displays:

    [WiFiEsp] Cannot initialize ESP module
    [WiFiEsp] >>> TIMEOUT >>>
    [WiFiEsp] No tag found
    WiFi shield not present

Milan991, 2020-10-16:
Same problem on my side:

```
[WiFiEsp] >>> TIMEOUT >>>
[WiFiEsp] >>> TIMEOUT >>>
[WiFiEsp] >>> TIMEOUT >>>
[WiFiEsp] Cannot initialize ESP module
```

mvturkes, 2020-11-07:
hello, i found a solution in my own way.
At first I changed the firmware of esp32. I compiled "esp32-at" with the directives here.
https://esp32.com/viewtopic.php?t=8821
https://www.esp32.com/viewtopic.php?t=15771
Then I made the following changes to the pins_arduino.h file.
C:\Users\username\AppData\Local\Arduino15\packages\Maixduino\hardware\k210\0.3.11\variants\sipeed_maixduino

#define MD_PIN_MAP(fpio) (((fpio) < 19) ? _maixduino_pin_map[(fpio)] : (fpio))

static const uint8_t _maixduino_pin_map[19] = {4, 5, 21, 22, 23, 24, 32, 15, 14, 13, 12, 11, 10, 3, 31, 30, **_6, 7_,** 16};

#define ORG_PIN_MAP(org_pin) _original_pin_map[(org_pin)]

static const uint8_t _original_pin_map[48] = {255, 255, 255, 13,  0,   1,   **_16, 17_,** 255, 255, 
                                              12,  11,  10,  9,   8,   7,   16, 255, 255, 255, 
                                              255, 2,   3,   4,   5,   255, 255, 255, 255, 255, 
                                              15,  14,  6,   255, 255, 255, 255, 255, 255, 255, 
                                              255, 255, 255, 255, 255, 255, 255, 255};

Then in the sample scannetworks.ino file.
  Serial1.begin(115200,**_16,17_**);

after these changes it works great.

canDry, 2021-02-28:
> hello, i found a solution in my own way.
> At first I changed the firmware of esp32. I compiled "esp32-at" with the directives here.
>....
> after these changes it works great.

Any chance you could post your compiled esp32 (for maxduino)?  :)

X-iv, 2021-07-04:
> I confirm this problem with a Maixduino. Running the ScanNetworks example just displays:
> 
> ```
> [WiFiEsp] Cannot initialize ESP module
> [WiFiEsp] >>> TIMEOUT >>>
> [WiFiEsp] No tag found
> WiFi shield not present
> ```

Same issue with me.

romainreignier, 2022-10-11:
The default firmware on the ESP32 uses SPI communication, see my comment here to see how to use it: https://github.com/sipeed/Maixduino/issues/87#issuecomment-1275044906