sipeed/Maixduino #77: Analogue inputs don't work

technoblogy, 2020-03-18:
The analogue inputs don't work; trying:

    Serial.println(analogRead(A0));

always prints 0, and:

    Serial.println(analogRead(33));

always prints 33.

CombiesGit, 2020-04-05:
None analog input pin exist!

See wiring.c
`uint32_t analogRead( uint32_t ulPin ) 
{
#if (defined(BOARD_SIPEED_MAIX_GO) || defined(BOARD_SIPEED_MAIX_ONE_DOCK) )
    return 0;
#endif
}`

technoblogy, 2020-04-05:
I'm referring to the Maixduino board. There are six analog pins; see here:

[image]

CombiesGit, 2020-04-05:
Ahh, 

The analog pins are owned by ESP not by K210.

technoblogy, 2020-04-05:
They still need to be supported in the Maixduino core.

barni2000, 2020-07-29:
For use analog pins you will need write SPI code for connection between the ESP32 and k210.Analog pins only works from the ESP.

technoblogy, 2020-07-29:
Is there a plan to add the necessary code to the Arduino core, so analogRead() will work as expected?