sipeed/Maixduino #83: Problem with package_Maixduino_k210_index.json

solariun, 2020-05-19:
Both Arduino json indexes are downloading HTML a informing it was permanently moved (HTTP 301). As a Result it is corrupting the lib for those whose already have it and not letting other download new ones.

    http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json
    http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_dl_cdn_index.json

Purmpol, 2020-05-20:
Yes, me too.
These were the error msg I got.

Skipping contributed index file /home/em/.arduino15/package_Maixduino_k210_dl_cdn_index.json, parsing error occured:
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('<' (code 60)): expected a valid value (number, String, array, object, 'true', 'false' or 'null')

Skipping contributed index file /home/em/.arduino15/package_Maixduino_k210_index.json, parsing error occured:
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('<' (code 60)): expected a valid value (number, String, array, object, 'true', 'false' or 'null')
 at [Source: (FileInputStream); line: 1, column: 2]

solariun, 2020-05-20:
Adding: on https, that will the same calling from the web view interface, it redirect for a https://dl.sipeed.com/limit, suggesting something regarding web traffic limits

solariun, 2020-05-20:
> Yes, me too.
> These were the error msg I got.
>
> Skipping contributed index file /home/em/.arduino15/package_Maixduino_k210_dl_cdn_index.json, parsing error occured: [...]

     This error is, actually, an HTML returned by the site with status HTTP 200 OK, but actually telling it was HTTP 301 (Moved permanently), leading Arduino to not be able to address the lib, platform and programmer... ;( I am suffering with the same problem.

JonRobo, 2020-05-20:
I'm having the same problem, and it's not just the json file but all files needed will have errors downloading.

solariun, 2020-05-21:
> I'm having the same problem, and it's not just the json file but all files needed will have errors downloading. I downloaded the json file manually from sipeed and then threw it up on github here
> https://raw.githubusercontent.com/JonRobo/Maixduino_Code/master/package_Maixduino_k210_dl_cdn_index.json
> 
> But then i get CRC mismatch errrors on the other downloads

What I am suspecting it is something related to 'Limits'. sometimes downloading it working, but most of the times it will not. I do not know sipeed's ISP contract, but I suspect they have limits... so the best way is to move those files to guithub

yulonglovecode, 2020-05-21:
any idea how to solve it ?

solariun, 2020-05-21:
> any idea how to solve it ?

I started by sending a direct main to the sipeed support, now I am trying to recreate the directories, but I will only have time in the weekend, if I succeed I will fork Maixduino project and add the json manually, this way, we all can use while they are fixing it. I also suggested if they could move the entire maixduino json, BSP and Arduino configurations entirely to GitHub, so, everyone could do it.  Ultimately I asked SiPeed to come here and tell us something, in the end of the day we are helping propagating their name. In my case I am creating a ultra lightweight cooperative thread lib capable to run virtually in anything, from software, to processors and microcontroller, using stack memory page backup and restorations, context switching and so on, but in a way it can run everywhere, even with freeRTOS. So I use sipeed boards because I use Arduino as my HAL (Hardware abstraction Layer) this way I can only focus on process control and it will do the trick making things compatibles, so, Arduino helped me also test with RISC-V processors with a lot o success.

Purmpol, 2020-05-22:
There's some progress. After adding the url http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_dl_cdn_index.json, I could see Maixduino(k210) in Boards Manager list.

However, when I was trying to install , it showed the error msg below:
---------------------------------------
 CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:105)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more

TridentTD, 2020-05-22:
Can't setup for Windows' Arduino IDE
How to solve it ?

[image]

solariun, 2020-05-22:
> Can't setup for Windows' Arduino IDE
> How to solve it ?
> 
> [image]

I even sent a email for supeed with that and suggesting to move it entirely to GitHub.

solariun, 2020-05-25:
Please. all, replace your link by this  https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?id=arduino 

Since I am not sure for how long it will be kept online, please disable your existing link (http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json) by adding '#' to the beginning of the line and add this one right after.

it will download it and install.

solariun, 2020-05-25:
But it is shame that Sipeed has not answered my emails or even come forth to explain what is happening.

yulonglovecode, 2020-05-25:
> Please. all, replace your link by this https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?id=arduino
>
> Since I am not sure for how long it will be kept online, please disable your [...]

the maxiduino (k210) appeared on the board manager.
However when i want to install it , there is a red warning "CRC doestn't match, file is corrupted. It may be a temporary problem, please retry later

solariun, 2020-05-25:
I closed by mistake

solariun, 2020-05-25:
> the maxiduino (k210) appeared on the board manager.
> However when i want to install it , there is a red warning "CRC doestn't match, file is corrupted. It may be a temporary problem, please retry later

I had a problem as well, so, I guess the best option is to clean up the MAIXDUINO directory in the Arduino15/packages and reinstall... try to backup it for security reasons and reinstall it using the link I found.  I already have mine running again, now applying my smoke test for my Universal Tread Library that also support RISC-V over its FreeRTOS or Non-OS.

JonRobo, 2020-05-25:
I removed anything pre-existing for maixduino in my Arduino15 directory and I'm still getting the CRC mismatch error when trying to download. The JSON file is okay, but still no success on the install.

solariun, 2020-05-25:
> I removed anything pre-existing for maixduino in my Arduino15 directory and I'm still getting the CRC mismatch error when trying to download. The JSON file is okay, but still no success on the install. 

Can you post the exactly error message?

Purmpol, 2020-05-26:
> I removed anything pre-existing for maixduino in my Arduino15 directory and I'm still getting the CRC mismatch error when trying to download. The JSON file is okay, but still no success on the install.

**Same here. No hope.**
------------------------------
 CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:105)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more

myuanz, 2020-05-26:
Sorry, it's OK now.

Give it a try!

JonRobo, 2020-05-26:
I tried this morning only to find that the download limit from sipeed.com had been reached again. There's probably a similar limit on the libraries to follow the json file.

[image]

However I did get mine to install this morning by posting a previously downloaded JSON file for Maixduin 0.3.11. You can use https://raw.githubusercontent.com/JonRobo/Maixduino_Code/master/package_Maixduino_k210_index.json

To at least guarantee the JSON file can be downloaded, the rest is up to chance for now until it is moved from sipeed to github.

myuanz, 2020-05-26:
> I tried this morning only to find that the download limit from sipeed.com had been reached again. There's probably a similar limit on the libraries to follow the json file.
>
> [image]
>
> However I did [...]

We removed the traffic limit, it come from an attack not long ago

solariun, 2020-05-26:
> We removed the traffic limit, it come from an attack not long ago

Are you using this: https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?id=arduino as well ?  It is working for me. IT you where saying that limit, probably you where not using the ?id=arduino

myuanz, 2020-05-26:
> Are you using this: https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?id=arduino as well ? It is working for me. IT you where saying that limit, probably you where not using the [...]

This issue, emmm, is actually a combination of many issues, so do not discuss it again. (I blushed

JonRobo, 2020-05-26:
> Are you using this: https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?id=arduino as well ? It is working for me. IT you where saying that limit, probably you where not using the [...]

I was using that link. The package is installed for me now. However now that I have Maxduino it's time to dig through all the other bugs such as snap_fail in the selfie example XD

Purmpol, 2020-05-27:
Today I can install it successfully. Thanks for every people who involved.

closed 2020-05-27 by myuanz