sipeed/Maixduino #84: How to solve Maix-Dock's uploading error on Arduino IDE

TridentTD, 2020-05-26:
I am a newbie for Maix Dock,
I am trying to upload any code to Maix Dock on Arduino IDE.

When I connect the Maix Dock to my computer,
my computer can find the Maix Dock's port at COM4.

However when I upload to the Maix Dock,
it can't be uploaded completely.
The error occurs as the following

[image]
****

How to solve the issue for uploading to the board?
Thank you.

TridentTD, 2020-05-27:
It's already solved!
By programmer to k-flash.

Birk-maker, 2020-10-21:
I just got a Maixduino and have exactly the same problem using the Arduino IDE as you - before you solved it.  The Arduino IDE is set to k-flash as the programmer setting, so I'm not sure why the uploading is not working.  How exactly did you solve it?  Do I need to install something else?
Thanks!