sipeed/Maixduino #85: "json file signature verification failed" on Arduino boards file

technoblogy, 2020-05-29:
The Arduino IDE boards file for using Maixduino on the Arduino IDE is currently giving an error:

> http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json file signature verification failed. File ignored.

This makes it impossible to develop programs with any of the Sipeed K210 RSIC-V boards, including the Maixduino, MAiX One Dock, and MAiX BiT.

Please can this be fixed as soon as possible, or alternatively explain how to fix it.

Thank you, David

technoblogy, 2020-05-29:
OK, for anyone else who's stuck with this problem I've found a workaround:

* Locate the previously working **Maixduino** folder; on a Mac it's in:

    /Users/david/Library/Arduino15/packages

where "david" will be whatever is your username.

* Locate the **k210** folder in the **hardware** folder inside this.

* Copy the **k210** folder and its contents to the **hardware** folder in your **Arduino** sketches folder.

When you quit and rerun the Arduino IDE the **Maixduino** section should now appear on the **Board** menu.

technoblogy, 2020-05-29:
As a workaround I was able to get it working again by getting the older package files off backup; these are:

    package_Maixduino_k210_index.json
    package_Maixduino_k210_index.json.sig.tmp

On a Mac they are in:

    /Users/david/Library/Arduino15

where david is whatever your username is.

However, due to the file signature verification failure they'll get deleted if you go into the **Boards Manager**.

technoblogy, 2020-07-14:
This is still a problem for me. Is anyone else getting this?

myuanz, 2020-07-15:
> This is still a problem for me. Is anyone else getting this?

I can't reproduce it, could you give the error message

technoblogy, 2020-07-15:
I'm using this boards file in Arduino **Preferences**:

    http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json

Every time I go into **Boards Manager...** I get the error:

    http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json file signature verification failed. File ignored.

and my existing Maixduino files are deleted:

    package_Maixduino_k210_index.json
    package_Maixduino_k210_index.json.sig.tmp

so the **Maixduino** entries disappear from the **Boards** menu.

I'm using Arduino 1.8.9 on a MacBook Pro with macOS High Sierra 10.13.6.

myuanz, 2020-07-15:
How about trying `http://cn.dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json`? 
BTW, could give the error message when you visit `http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json` in browser.

technoblogy, 2020-07-15:
> How about trying http://cn.dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json?

Same problem:

    http://cn.dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json file signature verification failed. File ignored.

> BTW, could you give the error message when you visit http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json in browser.

There's no error message, but the behaviour is slightly different from other boards files. For example, if I go to:

    https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json

it downloads a file `package_seeeduino_boards_index.json`, but when I go to:

    http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json

it downloads a file `package_Maixduino_k210_index.json.txt` - in other words, it adds a ".txt" extension.

myuanz, 2020-07-15:
I can not trigger your problem in 1.8.13. It seems that the old version can not deal HTTP 301. 

Try 1.8.13?

technoblogy, 2020-07-15:
OK I'll let you know.

technoblogy, 2020-07-15:
I've upgraded to 1.8.13 and it seems OK now - thanks!

closed 2020-07-15 by technoblogy