sipeed/Maixduino #87: Communication with the ESP32

bmitov, 2020-06-25:
I have not been able to find much information on this :-( .

1. Does the ESP32 come preprogrammed with AT firmware, or do I have to burn it separately.
2. What pins it is connected to? I have seen that is it connected to D 6 and 7 of the K210. Is this correct?
3. What is the serial speed by default for the AT firmware if it comes programmed out of the box?

chegewara, 2020-06-25:
I have not been using it for almost a year, so i forgot most of it, but afaik its using SPI, and esp32 should be (at least it was) programmed with fw that allows communication. More info you may find here:
https://en.bbs.sipeed.com/

bmitov, 2020-06-25:
Thank you @chegewara . What Chip Select is used to talk to the ESP32 from the K210 ?
Do you happen to have some example code, or link to tutorial or something?

chegewara, 2020-06-25:
This is python code, but i hpe it will help you:
https://github.com/chegewara/sipeed-maixduino-test-apps/blob/master/platformIO/MaixPy/esp32-wifi.py

chegewara, 2020-06-25:
You can also flash esp32 with your own FW. Here is very simple test that is using UART to communicate k210 with esp32.
esp32 is flashed with custom fw:
https://github.com/chegewara/sipeed-maixduino-test-apps/tree/master/platformIO/freertos/uart-to-esp32

btw, esp32 is pre-programmed with NINA firmware

bmitov, 2020-06-25:
@chegewara I know that I can flash the ESP32 :-) I can even program it with Visuino . It is already supported. I did not want to override any factory code however. My question was what is on it factory ready? I wanted to test it before I override the code ;-)

bmitov, 2020-06-25:
In essence if the ESP32 comes pre programmed with AT type firmware, then I can add the AT support on the K210 side in Visuino as a WiFi module. Otherwise, I will just leave it as Serial and SPI channels for the ESP32 and just allow boths sides to be programmed individually.

chegewara, 2020-06-25:
Like i mentioned in other topic, its NINA.
https://github.com/sipeed/Maixduino_esp32_fimware

bmitov, 2020-06-25:
@chegewara Thank you so much! :-)
It is programmed when it arrives right :-) I will test shortly.

mattuna15, 2021-01-12:
If you want to use UART to send AT commands to the ESP32 you have to make the following changes.
Download the Espressif AT firmware and change this file:

**components/customized_partitions/raw_data/factory_param/factory_param_data.csv**
```
-PLATFORM_ESP32,WROOM-32,,0xfcfc,2,1,78,1,1,13,CN,115200,17,16,15,14,-1,-1
+PLATFORM_ESP32,WROOM-32,,0xfcfc,2,1,78,0,1,13,CN,115200,1,3,-1,-1,-1,-1
```

Then in the Arduino library, changes the pin mapping for pins 6 & 7 to k210 pins 6 & 7

Arduino15/packages/Maixduino/hardware/k210/0.3.11/variants/sipeed_maixduino/pins_arduino.h

`static const uint8_t _maixduino_pin_map[17] = {4, 5, 21, 22, 23, 24, **6, 7,** 14, 13, 12, 11, 10, 3, 31, 30, 16};`

Using these changes I can now use the internal UART between the k210 and ESP32 for AT commands. You can also still use the ESP32 directly.

ridvan70, 2021-11-21:
> @chegewara I know that I can flash the ESP32 :-) I can even program it with Visuino . It is already supported. I did not want to override any factory code however. My question was what is on it factory ready? I wanted to test it before I override the code ;-)

I get the following information from a python script:

**[esp32_spi] use soft spi
1.4.0**

I bought mine in the first quarter of 2021.

There is new firmware by Sipeed, but have succeeded to flash it, yet.

romainreignier, 2022-10-11:
From the NINA source code, the ESP32 communicates through SPI and not a serial port with AT commands.
I have managed to communicate it with the [fork of WiFiEsp](https://github.com/scpcom/WiFiEsp/tree/esp32_spi) (**branch esp32_spi**) by @scpcom and using its [fork of Maixduino](https://github.com/scpcom/Maixduino).

Embedd-sva, 2024-09-15:
Hello guys I want program to establish communication between k210 and esp32 using platformio