sipeed/Maixduino #92: Support for the Sipeed Maix Amigo

rei-vilo, 2020-09-16:
Any support for the [Sipeed Maix Amigo](https://www.seeedstudio.com/Sipeed-Maix-Amigo-p-4689.html) board?

kallisti5, 2020-11-01:
Yup.. looking for Sipeed Maix Amigo support as well

kallisti5, 2020-11-01:
The schematic is here:
https://dl.sipeed.com/MAIX/HDK/Sipeed-Amigo

I started working on the pin map for Maixduino based on it... but not super clear.

```c
#ifndef _VARIANT_BOARD_SIPEED_MAIX_GO
#define _VARIANT_BOARD_SIPEED_MAIX_GO

#include <stdint.h>

#define RISCV
#include "platform.h"

#include "Arduino.h"
#include "pwm.h"

#ifdef __cplusplus
#include "UARTClass.h"
extern class UARTHSClass Serial;
extern class UARTClass Serial1;
extern class UARTClass Serial2;
extern class UARTClass Serial3;
#endif

/* BOARD  PIN DEFINE */
/* LEDs */
#define PIN_LED_GREEN        15
#define PIN_LED_BLUE         17
#define PIN_LED_RED          14
#define PIN_LED              32
#define LED_BUILTIN          32
#define LED_GREEN            15
#define LED_BLUE             17
#define LED_RED              14

/* KEY */
#define KEY0                  9 // ? STM32
#define PIN_KEY_DOWN         20
#define PIN_KEY_UP           23
#define PIN_KEY_PRESS         9 // ? STM32

// NO IDEA ON THESE. Values from Maix Go
#if 0

/* MIC ARRAY */
#define MIC_BCK              18
#define MIC_WS               19
#define MIC_DAT3             20
#define MIC_DAT2             21
#define MIC_DAT1             22
#define MIC_DAT0             23
#define MIC_LED_DAT          24
/* SPI0 */
#define SPI0_CS1             25
#define SPI0_MISO            26
#define SPI0_SCLK            27
#define SPI0_MOSI            28
#define SPI0_CS0             29

/* I2S */
#define I2S_WS               33
#define I2S_DA               34
#define I2S_BCK              35
/* LCD */
#define LCD_CS               36
#define LCD_RST              37
#define LCD_DC               38
#define LCD_WR               39

#endif

// U2 UART
#define RX0                   9
#define TX0                   7

// U3 UART
#define RX1                  25
#define TX1                  22

// U1 UART (ISP)
#define RX2                   4
#define TX2                   5

#define SDA                  27
#define SCL                  24

#define MD_PIN_MAP(fpio)   (fpio)
#define ORG_PIN_MAP(org_pin)    (org_pin)

static const uint8_t SS   = SPI0_CS0 ;
static const uint8_t MOSI = SPI0_MOSI;
static const uint8_t MISO = SPI0_MISO;
static const uint8_t SCK  = SPI0_SCLK;

typedef struct _pwm_fpio_set_t{
    pwm_channel_number_t channel;
    pwm_device_number_t device;
    uint8_t inuse;
}pwm_fpio_set_t;

#define VARIANT_NUM_GPIOHS (32)
#define VARIANT_NUM_GPIO   ( 8)
#define VARIANT_NUM_PWM    (12)
#define VARIANT_NUM_I2C    ( 3)
#define VARIANT_NUM_SPI    ( 3)
#define VARIANT_NUM_UART   ( 3)

#endif
```

kalon33, 2021-10-20:
With latest git snapshot, I got these errors trying to compile for Maix Amigo:

```
Arduino: 1.8.13 (Linux), Board: "Sipeed Maix Amigo, Default, 400MHz CPU Clock Frequency, 2 Mbps, open-ec"
/home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/hardware -hardware /home/nicolas/.arduino15/packages -hardware /home/nicolas/Arduino/hardware -tools /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/tools-builder -tools /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/hardware/tools/avr -tools /home/nicolas/.arduino15/packages -built-in-libraries /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/libraries -libraries /home/nicolas/Arduino/libraries -fqbn=Maixduino:cores:amigo:toolsloc=default,clksrc=400,burn_baudrate=2000000,burn_tool_firmware=amigo -ide-version=10813 -build-path /tmp/arduino_build_66634 -warnings=none -build-cache /tmp/arduino_cache_857317 -prefs=build.warn_data_percentage=75 -verbose /home/nicolas/Arduino/Wio_Terminal_TF-MICRO_Weather_Prediction_test_static/Wio_Terminal_TF-MICRO_Weather_Prediction_test_static.ino
/home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/hardware -hardware /home/nicolas/.arduino15/packages -hardware /home/nicolas/Arduino/hardware -tools /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/tools-builder -tools /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/hardware/tools/avr -tools /home/nicolas/.arduino15/packages -built-in-libraries /home/nicolas/Téléchargements/arduino-1.8.13-linux64/arduino-1.8.13/libraries -libraries /home/nicolas/Arduino/libraries -fqbn=Maixduino:cores:amigo:toolsloc=default,clksrc=400,burn_baudrate=2000000,burn_tool_firmware=amigo -ide-version=10813 -build-path /tmp/arduino_build_66634 -warnings=none -build-cache /tmp/arduino_cache_857317 -prefs=build.warn_data_percentage=75 -verbose /home/nicolas/Arduino/Wio_Terminal_TF-MICRO_Weather_Prediction_test_static/Wio_Terminal_TF-MICRO_Weather_Prediction_test_static.ino
Using board 'amigo' from platform in folder: /home/nicolas/Arduino/hardware/Maixduino/cores
Using core 'arduino' from platform in folder: /home/nicolas/Arduino/hardware/Maixduino/cores
Detecting libraries used...
recipe.preproc.macros pattern is missing

Error compiling for board Sipeed Maix Amigo.
```

Please help

NameStream, 2023-09-17:
su09-17-2023

ImportError: cannot import name MIC_ARRAY

MAix Amigo

(1.) Hi, there.

(2.) Using MAixPy IDE, receiving error code for MAix Amigo:

ImportError: cannot import name MIC_ARRAY

(3.) MAix AMIGO

(3a.) SP-MOD MicArray Module

(3b.) Connector:  10P 0.5mm FPC

(3c.) Sipeed MAix R6 + 1 Microphone Array Board

(3d.) Located inside the back of the developer board #2970, moved the switch to M1N instead of STM.
* * * * * * * * {  By the way, STM yields same results. } * * * * * * * *

(3e.) MAixPy IDE:  Tools > Select Board > Sipeed Maixduino

(3f.) Will only run the code for “helloworld.py”, although the LCD image orientation is misaligned, upside down; the picture is smaller than the LCD screen; but image has mostly clear resolution without any noticeable lag, latency, delay.

(3g.) Receiving “ImportError:  cannot import name MIC_ARRAY”
when attempting to run the code for:

ROUTINE Sound Source Localization

https://wiki.sipeed.com/soft/maixpy/en/develop_kit_board/module_microphone.html

*******and*******also*******for*******

AirPocket Soundman’s Sound Localizer
https://github.com/airpocket-soundman/SoundVisualizer/blob/main/src/mic_array_en_ver204.py

(3h.) Pin assignments used for MAix AMIGO

LED_CLK_GPIOHS28…….23

LED_DAT_GPIOHS27…….20

DAT 0…………….......……12

DAT 1……………..…..…….…9

DAT 2………………..…...……31

DAT 3………….……..……...7

WS……………..…....…..……19

BCK_SCK……...…..…………17

(4.) Would you list the correct pin assignments for the:

(4a.) MAix Amigo #2970 in the M1n position

(4b.) MAix Amigo #2970 in the STM position

(5.) Thanks.

(6.) Toodles!