sipeed/Maixduino #97: Something wrong with I2C

Kabron287, 2020-11-01:
I'm trying to get working **MLX90640** camera with **MAIX GO** board.
**I2C** pins are **30 - SCL, 31 - SDA**. Attached example reads **MLX** serial number.
As could be seen from the attached diagrams, transmit is OK, but received data wrong. 
So, image and all other datas are corrupted. 
As ethalon exchange with M5Stack attached.
Correct **SN** is **0608 A233 018**5
[image]
[image]
[Test_MLX.zip](https://github.com/sipeed/Maixduino/files/5470471/Test_MLX.zip)

Kabron287, 2020-11-02:
When I change to **SlowSoftI2C** driver, normal communication RESTORED.
It is a definite BUG in I2C driver!

LFerOrtiz, 2022-03-09:
I have the same problem.
The program read the EEPROM data from the sensor, but the data is corrupted.