sipeed/Maixduino #100: the latest version cannot download from Arduino boardmanager

ml2068, 2020-12-21:
when I try to use Arduino boardmanager download the 0.3.11version，erro showup :"CRC校验出错：文件可能已损坏。这可能是个暂时性的问题，请稍后重试"

zaferAt, 2020-12-22:
Same thing happened to me. This is because "dl.sipeed.com" address it is trying to blocks large files being downloaded without captcha.

You can workaround this issue by downloading the required large file manually from here "https://github.com/kendryte/kendryte-gnu-toolchain/releases" and move it under "\staging\packages" folder of your Arduino installation. Wherever located under your operating system. In my case it was "C:\Users\\{username}\AppData\Local\Arduino15\staging\packages" folder and I needed "kendryte-toolchain-win-amd64-8.2.0-20190409.zip" file. This is probably what you need for windows 0.3.11 version.

ml2068, 2020-12-23:
ok, I 'll try it. thanks !

closed 2020-12-23 by ml2068

Neutree, 2020-12-23:
@ml2068  link fixed, you can try again

Neutree, 2020-12-23:
the error occurs by dl.sipeed.com's update, now it fixed