sipeed/Maixduino #105: Out of normal MicroPython Heap Memory!

diazGT94, 2021-01-19:
Hello,

I'm running a script that imports a video from the SD and I'm using the face model to track the face in the video. However, when I want to draw a rectangle in the face I get the following error: 

Out of normal MicroPython Heap Memory! Please reduce the resolution of the image you are running this algorithm on to bypass this issue!

How can I fix it?

BackMountainDevil, 2022-03-23:
[ MemoryError: Out of normal MicroPython Heap Memory! Please reduce the resolution of the image you are running this algorithm on to bypass this issue! #393 ](https://github.com/sipeed/MaixPy/issues/393 ) You close it. But I do not see the solution.

LsDevLab, 2022-09-17:
Solution here [MemoryError: Out of normal MicroPython Heap Memory! Please reduce the resolution of the image you are running this algorithm on to bypass this issue! #393 ](https://github.com/sipeed/Maixduino/issues/105#:~:text=on%2023%20Mar-,MemoryError%3A%20Out%20of%20normal%20MicroPython%20Heap%20Memory!%20Please%20reduce%20the%20resolution%20of%20the%20image%20you%20are%20running%20this%20algorithm%20on%20to%20bypass%20this%20issue!%20%23393,-You%20close%20it)

BackMountainDevil, 2022-09-17:
@LsDevLab Thanks you for your reply.