sipeed/Maixduino #106: MobileNet example is not working with Sipeed_GC0328

TA-Robot, 2021-01-20:
I tried to use working mobilenet example with Sipeed_GC0328. But it's not working.
When mbnet.show() function called, it clashed with core dump like below.

Please fix this or give me example with Sipeed_GC0328!

`core dump: misaligned load
Cause 0x0000000000000004, EPC 0x0000000080000b5e
reg[00](zero ) = 0x0000000000000002, reg[01](ra   ) = 0x0000000080000d5c
reg[02](sp   ) = 0x0000000080066e70, reg[03](gp   ) = 0x0000000080030ae0
reg[04](tp   ) = 0x000000008005f100, reg[05](t0   ) = 0x0000000000000000
reg[06](t1   ) = 0x000000008000639a, reg[07](t2   ) = 0x0000000000000000
reg[08](s0/fp) = 0x00000000800306f0, reg[09](s1   ) = 0x0000000080030760
reg[10](a0   ) = 0x000000000000000d, reg[11](a1   ) = 0x000000000000001a
reg[12](a2   ) = 0x000000000000000b, reg[13](a3   ) = 0x000000000000000b
reg[14](a4   ) = 0x0000000038000000, reg[15](a5   ) = 0x000000000000000a
reg[16](a6   ) = 0x0000000000001000, reg[17](a7   ) = 0x0000000000000003
reg[18](s2   ) = 0x0000000080030720, reg[19](s3   ) = 0x00000000800306b0
reg[20](s4   ) = 0x0000000000010000, reg[21](s5   ) = 0x0000000080080d70
reg[22](s6   ) = 0x00000000800055bc, reg[23](s7   ) = 0x0000000000000000
reg[24](s8   ) = 0x0000000000000000, reg[25](s9   ) = 0x0000000000000000
reg[26](s10  ) = 0x0000000000000000, reg[27](s11  ) = 0x0000000000000000
reg[28](t3   ) = 0xfffffffffffffff1, reg[29](t4   ) = 0x0000000050000300
reg[30](t5   ) = 0x0000300000000000, reg[31](t6   ) = 0x0000018000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028630()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028640()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028650()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028660()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028688()
freg[10](fa0 ) = 0x00000000bd20a490(), freg[-1610612736]() = 0x00000000800286b8()
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800286e0()
freg[14](fa4 ) = 0x000000003f8dd143(), freg[-1610612736]() = 0x00000000800286f0()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028700()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028728()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028738()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028748()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028758()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x0000000080028768()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x0000000080028790()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x00000000800287a0()`

TA-Robot, 2021-01-20:
I don't know I totally solved this problem, but now my program working!

In my case, above core dump gone after adding "return 0;" to last of  int MBNet_1000::show() function.

this make so good.

abhi-84, 2021-11-25:
getting similar issue, could not identify the source of the problem.
MBNet_1000:show() function have return type void in example program so return 0 is not required.