sipeed/Maixduino #118: Cannot install board definitions - wrong link and wrong SHA256

AIWintermuteAI, 2021-11-29:
The board definitions cannot be installed for at lest two reasons:
1) https://github.com/UT2UH/Maixduino/blob/gh-pages/package_Maixduino_boards_index.json
This leads to a Github page with package_Maixduino_boards_index.json - and not the file content itself. What you need is 
https://raw.githubusercontent.com/UT2UH/Maixduino/gh-pages/package_Maixduino_boards_index.json
2) Unfortunately there is a very clear mix-up about board definitions version and SHA256 checksum. 
[image]
It downloads the Maixduino-0.3.11.zip, but specifies wrong filename and version. Additionally, the checksum for Maixduino-0.3.11.zip is
[image]
Very clearly not the same is specified in package_Maixduino_boards_index.json, which leads to failed installation.

Proposed solution:
1) Make a new release in https://github.com/sipeed/Maixduino repository, which include package_Maixduino_boards_index.json and Maixduino-0.3.12.zip (increment version by 0.0.1, don't change the content). 
2) Make sure links and checksums are correct.

Thanks!

Neutree, 2021-11-29:
the third party package's error, change to official should be ok https://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json

AIWintermuteAI, 2021-11-29:
Thanks! I'll test the installation tonight and if everything is fine, close the issue.

AIWintermuteAI, 2021-12-13:
Everything seems to be in order now. Closing the issue.

closed 2021-12-13 by AIWintermuteAI