sipeed/Maixduino #119: no serial connection

Matritze, 2022-03-01:
Hey Maixfriends,

my problem is that I don't get a serial connection to my Maixduino anymore.
After unsuccessful trying to upload a model into flash my display become white and up to this moment I cannot get any serial connection anymore. Flashing the firmware via kflash is not possible. After a few seconds an error message appears. During this time only the R-K210 LED is flashing, but not the T-K210. So I think the K210 does not send an answer via serial. Is there any other possibility to flash the FW? I fear that the firmware has been killed somehow.

voidstring, 2022-03-08:
can you share the error message ?

Matritze, 2022-03-08:
I will share errormessage later.

Matritze, 2022-03-08:
[image]
That is the errormessage from kflash. The com-port was detected correctly before.

voidstring, 2022-03-08:
On maixduino com3? Are you sure ?
[image]
usb serial port you should choose ? (foregoing example port 5 or port 6)

Matritze, 2022-03-09:
The Port was 16. Think the error Massage displays always the same. Even on an other Computer with different Port the Massage was the same. The Port is not the problem. There comes no answer from the k210.

voidstring, 2022-03-23:
I understood you. But I don't know how to solve it. Maybe even unsolvable.