sipeed/Maixduino #120: [TCP/İP Socket] live stream using maixduino as client

voidstring, 2022-03-08:
My Code:
`
SSID = "MW40V"
PASW = "***"

import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

def enable_esp32():
    from network_esp32 import wifi
    if wifi.isconnected() == False:
        for i in range(5):
            try:
                # Running within 3 seconds of power-up can cause an SD load error
                # wifi.reset(is_hard=False)
                wifi.reset(is_hard=True)
                print('try AT connect wifi...')
                wifi.connect(SSID, PASW)
                if wifi.isconnected():
                    break
            except Exception as e:
                print(e)
    print('network state:', wifi.isconnected(), wifi.ifconfig())

enable_esp32()

def enable_espat():
    from network_espat import wifi
    if wifi.isconnected() == False:
        for i in range(5):
            try:
                # Running within 3 seconds of power-up can cause an SD load error
                # wifi.reset(is_hard=False)
                wifi.reset()
                print('try AT connect wifi...')
                wifi.connect(SSID, PASW)
                if wifi.isconnected():
                    break
            except Exception as e:
                print(e)
    print('network state:', wifi.isconnected(), wifi.ifconfig())

#enable_espat()

def network_wiznet5k():
    from network_wiznet5k import lan
    from machine import SPI
    from Maix import GPIO
    if lan.isconnected() == False:
        WIZNET5K_SPI_SCK = 21
        WIZNET5K_SPI_MOSI = 8
        WIZNET5K_SPI_MISO = 15
        WIZNET5K_SPI_CS = 20
        spi1 = SPI(4, mode=SPI.MODE_MASTER, baudrate=600 * 1000,
                    polarity=0, phase=0, bits=8, firstbit=SPI.MSB, sck=WIZNET5K_SPI_SCK, mosi=WIZNET5K_SPI_MOSI, miso=WIZNET5K_SPI_MISO)
        for i in range(5):
            try:
                lan.reset(spi1, WIZNET5K_SPI_CS)
                print('try connect lan...')
                if lan.isconnected():
                    break
            except Exception as e:
                print(e)
    print('network state:', lan.isconnected(), lan.ifconfig())

# network_wiznet5k()

import socket

ADDR = ("192.168.1.***", 23***)

sock = socket.socket()
sock.connect(ADDR)
sock.settimeout(1)
clock = time.clock()
while 1:
    sock.send(sensor.get_fb())
    #data = sock.recv(10) # old maxipy have bug (recv timeout no return last data)
    #print(data) # fix
    try:
      data = b""
      while True:
        tmp = sock.recv(1)
        print(tmp)
        if len(tmp) == 0:
            raise Exception('timeout or disconnected')
        data += tmp
    except Exception as e:
      print("rcv:", len(data), data)
    #time.sleep(2)

sock.close()
`
What type of data should I send? I keep getting errors.(İMG CONVERT ??? Sensor.Snapshot() etc..)

**Err Msg:**
"Traceback (most recent call last):
  File "<stdin>", line 82, in <module>
OSError: [Errno 5] EIO"

voidstring, 2022-03-12:
i am fixed.

closed 2022-03-12 by voidstring

BackMountainDevil, 2022-03-25:
Helllo, have you met this error - ImportError: no module named 'network_esp32'? I try to run your code or  [network demo](https://wiki.sipeed.com/soft/maixpy/zh/course/network/network_config.html?highlight=wifi#%E5%A6%82%E4%BD%95%E8%81%94%E7%BD%91%EF%BC%9F) will got this error.

my info:
- maxiduino 固件：maixpy_v0.6.2_73_g1a4f278a5.bin
- esp32 固件：maixduino_esp32_firmware_v1.4.0.bin
- 错误：`ImportError: no module named 'network_esp32'`
- 自行编译固件 模块配置 中无网络、esp32模块，社区搜索错误相关问题
- wiki [代码](https://wiki.sipeed.com/soft/maixpy/zh/course/network/network_config.html?highlight=wifi#%E5%A6%82%E4%BD%95%E8%81%94%E7%BD%91%EF%BC%9F)如下

```python
import network_esp32
print(network_esp32)
print(dir(network_esp32))

from network_esp32 import wifi
print(wifi)
```

voidstring, 2022-03-25:
Hi, 
I'm telling you what to do in order.

-  You have to open Maixpy İDE or OpenMv İDE.
- Connect  port using a board
[image]
- Send module network_esp32 to board