sipeed/Maixduino #121: Maixduino Operating system

mrynotfound, 2022-03-23:
Hello,

I am  gonna use sipeed maixduino, but i don't know that should i load an operating system, because if i should do that i wanna load linux operating system and i don't know how to do it. And i wanna run Scade display and Scade suite codes, how can i do that either? 

Thank you for your response,

voidstring, 2022-03-25:
I think it can be installed in v2(Maix V2)

romainreignier, 2022-10-11:
It is possible to use Linux on K210 based boards.
But note that this is a `no-MMU` Linux so it is not as easy to use than on a Raspberry Pi for example.
The process is quite easy thanks of [Buildroot](https://buildroot.org/), following [this README for the Maixduino](https://git.busybox.net/buildroot/tree/board/sipeed/maixduino/readme.txt?id=e381fa9cf21f4384ffd6b305829417e9d2dd4d94).