sipeed/Maixduino #122: [UART] Maixduino Reading Serial port

NikolaiBirolini, 2022-03-24:
Hello everyone !

I want to read my UART's output of my barcode scanner and print it on my computer.
My code : 

char outBarcode;

void setup()
{
    Serial.begin(115200);    
    Serial2.begin(115200, 6, 7);
}

void loop()
{
   Serial.println(Serial2.available());
   if (Serial.available()) 
     {
        Serial.println("Scanning")
        outBarcode = Serial2.read()
        delay(1000);
        Serial.println(out)
     } 
}

But my Serial2.available is always equal to 0 ....
So, firstly, is this code seems correct ?
And can I write others pins in Serial2.begin(baud, RX, TX) ? For example, RX = 10 and TX = 11 ?

I wish you all a good day !
Thanks for reading me !

voidstring, 2022-03-25:
Hi friend,
i will help you as far as i know.First, make sure you have made the connections correctly.Because I had such a problem.
[image]

You probably got the connection on the board wrong.If not, I'll try to help.

NikolaiBirolini, 2022-03-25:
Thank you my friend ! It was that !
Have a nice day !

closed 2022-03-25 by NikolaiBirolini