sipeed/Maixduino #126: How to use C/C++ PlatformIO/Arduino Framework  (Not MaixPy) to use wifi

marcusobrien, 2022-10-11:
Hi, 

How can I use the ESP32 Wifi module from C/C++ ? It only seems to work using the MaixPy wifi example. When I run the default ESP32 firmware, and this C++ application code

` Serial.begin(115200);   // initialize serial for debugging
  WiFi.init();    // initialize ESP32 WiFi module

  if (WiFi.status() == WL_NO_SHIELD) 
  {
    Serial.println("No WiFi HW - Halting FW");
    while (true);
  }

  int8_t numberNetworks = WiFi.scanNetworks();
  Serial.print("Scan Wifi Found : ");
  Serial.println(numberNetworks);`

This just prints the following output....

Error 4
Scan Wifi Found: 0

Note that the wifi works if I use MaixPy firmware and the example code. Also I have 2 Maixduinos both of them dont work with the C++ SDK, but both work with the Python way.

Where is the ESP32 FW that will allow me to use the wifi from the Maixduino C/C++ application ?

romainreignier, 2022-10-11:
See my comment https://github.com/sipeed/Maixduino/issues/87#issuecomment-1275044906

marcusobrien, 2022-10-13:
> See my comment [#87 (comment)](https://github.com/sipeed/Maixduino/issues/87#issuecomment-1275044906)

Hi - So which firmware do I put on the ESP32 on the Maixduino ? My ESP32 doesnt have the original  firmware anymore - as I've been experimenting with lots of different ones - there are 4 prebuilt binaries in the Maixduino web portal - do I sue one of those ? ie [https://dl.sipeed.com/shareURL/MAIX/factory_firmware/esp32](url)

Has the following
maixduino_esp32_firmware_v1.4.1_0x0.bin
maixduino_esp32_firmware_v1.4.0.bin
NINA_W122_33.bin
NINA_W122_25.bin

"I have managed to communicate it with the [fork of WiFiEsp](https://github.com/scpcom/WiFiEsp/tree/esp32_spi) (branch esp32_spi) by @scpcom and using its [fork of Maixduino](https://github.com/scpcom/Maixduino)."

Then you mention you have used two repos to talk to the Wifi, I'm not sure how I get this working. I have some demo code (as above), but once I have the above FW on the ESP32, do I just use this repo as the platform io library (branch esp32_spi) ?

https://github.com/scpcom/WiFiEsp/tree/esp32_spi

Or this one - the fork you mentioned ?

https://github.com/scpcom/Maixduino

scpcom, 2022-10-15:
Yes, my fork of WifiEsp is integrated into my fork of Maixduino:
https://github.com/scpcom/Maixduino/releases
It may work standalone but using my board package is the easiest way.
PS: If the package is installed an the board is selected in the Arduino IDE you can open the Tools menu and change the new option "WiFi Driver" to SPI