sipeed/Maixduino #141: RGB565 have inverted channels for display

dimitre, 2024-06-14:
I've noticed today that display uses wrong colors for RGB565

this is Blue ``` 0b111111 << 10 ```
this is Red ```0b11111 << 5```
and this is Green ```0b11111```

So channels are inverted and bit depth also, the correct order would be RGB with Green with 6 bits instead of blue.
Thank you

dimitre, 2024-06-14:
it seems to work OK on display if bits are swapped like this
```c++
uint16_t rgbto565(uint8_t r, uint8_t g, uint8_t b) {
	return ((b >> 3) << 11) | ((r>>3) << 6) | g >> 2;
}
```
instead of normal RGB565 bit order:
```c++
uint16_t rgbto5651(uint8_t r, uint8_t g, uint8_t b) {
	return ((b>>3) << 11) | ((g>>2) << 5) | r >> 3;
}
```

Neutree, 2024-06-15:
we only set invert rgb to bgr on maixcube maixamigo, but maixduino seems no need to invert anything.(it's benn long time last I play k210, so I can not remember it clearly now, you can find code in MaixPy-v1 repo)

https://github.com/sipeed/MaixPy-v1/blob/a95e2eee386a34c99fd990394fcbba19f149e149/components/drivers/lcd/src/lcd_rgb.c#L510

dimitre, 2024-06-15:
Now I understand better, if I draw to the LCD, RGB565 is correct
but if I draw to a canvas GFXCanvas16, then COLOR_RED draws blue.